LibreSolar/bms-5s50-sc #5: High-side switching

condorxan, 2017-07-28:
Hi, I'm also doing similar work. Came upon your work. It's very well done. However, I wanted to know is there any special reason why you've picked high-side switching? 
Regards, Orkhan.

martinjaeger, 2017-07-31:
Hi Orkhan, thanks for the feedback.
HS switching is necessary because of the communication interface. If the GND between two communicating devices is interrupted, the GND on one side is floating. So the voltage at the communication interface (here: CAN bus) could be pulled up to the battery voltage and get damaged.
In addition to that, it makes sure that you can always switch off safely in a system with GND connected to earth or a common potential (e.g. in a car).

condorxan, 2017-07-31:
Thanks for the clarification. 
For small solar storage project do you think HS switching is feasible, or should I stick with LS switching for now?
-- Orkhan.

scttnlsn, 2018-01-27:
Could some sort of digital isolator be used for the communication lines instead?  Could potentially be a little simpler to implement.  I'd imagine if you had an AC inverter hooked up to the load side of the low-side FETs you'd want that part earth grounded but is it necessary to earth ground the battery side as well?

The HS switching certainly makes for a more elegant solution here.  I'm surprised the reference implementation in the BQ769x0 datasheet does not show this approach.