LibreSolar/bms-5s50-sc #7: Wrong commit id for library submodule

goganchic, 2018-04-14:
```bash
clone --recursive https://github.com/LibreSolar/BMS-5s.git
Cloning into 'BMS-5s'...
remote: Counting objects: 108, done.
remote: Total 108 (delta 0), reused 0 (delta 0), pack-reused 108
Receiving objects: 100% (108/108), 3.51 MiB | 729.00 KiB/s, done.
Resolving deltas: 100% (50/50), done.
Checking connectivity... done.
Submodule 'library' (https://github.com/LibreSolar/KiCad-symbols) registered for path 'library'
Cloning into '/workspace/BMS-5s/library'...
error: no such remote ref 7d79f981fbde7c0778b5b1ace3a8dd21f3bbf14d
Fetched in submodule path 'library', but it did not contain 7d79f981fbde7c0778b5b1ace3a8dd21f3bbf14d. Direct fetching of that commit failed.
```

martinjaeger, 2018-04-15:
Thanks for the report. Will try to fix it next week.

martinjaeger, 2018-05-11:
Should be solved now (submodule removed and library directly saved in the repository).

closed 2018-05-11 by martinjaeger

goganchic, 2018-05-11:
Error disappeared, thanks!