LibreSolar/bms-5s50-sc #8: control-pcb layout fails DRC

johnfwhitmore, 2018-10-14:
To try this out I tried to order the board from jlcpcb, but they didn't like it, for some reason they didn't elaborate on. I've now looked at the project in kicad and got numerous issues when I ran a DRC check on the layout. Perhaps it's due to my version of Kicad so I'll go and update that... No updated and still throwing up a lot of errors.

martinjaeger, 2018-10-15:
Ok, I checked what was the issue. Somehow the Fab layers were deactivated which threw >1500 DRC errors. Might be because the design was done with the unstable KiCad V5 branch.

Anyway, corrected the problem and DRC should now run sucessfully on KiCad 5.

Please test again and give feedback. Thanks.

johnfwhitmore, 2018-10-16:
Thank you. Yes that has sorted out the issue and it looks like I could order that board from jlcpcb to try it out. I am however having a strange problem with the power part of the BMS design. I'm not sure if it's the output from Kicad or it's a jlcpcb problem.

To order from jlcpcb I normally just plot the pcb from Kicad and zip up the resulting files from the plot and drill files. I upload that zip file to jlcpcb and they process the zip file and come back with a graphical representation of the file and the size. I'll add an image of the control PCB of the design as an example. So whilst the control pcb works the power pcb just can't be processes by jlcpcb. I don't see a difference in how I'm plotting the two designs.

[image]

martinjaeger, 2018-10-16:
Do they also accept gerber? If yes, you could try to export gerber first.

Or you order at aisler.net. They produce very good quality PCBs (made in Germany) and also ship worldwide.

johnfwhitmore, 2018-10-16:
Oops, Oops Oops!!! Sorry St. Pauli my bad, I always plot from kicad and it defaults to "Gerber" so I never looked at the format. The power board is set to export an SVG format. No wonder jlcpcb didn't like that. I never even noticed that there were other formats, everything always defaulted to Gerber.

I should now go through the BOMs and see can I get all these components. Thanks for your help

johnfwhitmore, 2018-10-16:
One question about the components. The big Wurth connectors for the power, I don't see how you attach to them. It says they have a 9.7mm hole but there's no thread size listed?

https://ie.farnell.com/wurth-elektronik/74650195r/terminal-bush-thr-pcb-9-7mm-m5/dp/2748951

johnfwhitmore, 2018-10-16:
Actually looking at the cost I might be finding a different connector.

martinjaeger, 2018-10-16:
See here:
https://katalog.we-online.com/en/em/WP-THRBU_THROUGH-HOLE

They are cheaper if ordered directly from Würth.

closed 2018-10-20 by martinjaeger